pyvisa/pyvisa-sim #4: "error: package directory 'pyvisa-sim/testsuite' does not exist"

cmarqu, 2015-02-09:
Hi,

I'm getting the following error (I assume it's not critical, but wanted to report it nevertheless):

$ pip3.4 install --user --upgrade -v  https://github.com/hgrecco/pyvisa-sim/zipball/master
Downloading/unpacking https://github.com/hgrecco/pyvisa-sim/zipball/master
  Downloading master (unknown size): 
  Downloading from URL https://github.com/hgrecco/pyvisa-sim/zipball/master
...Downloading master (unknown size): 19kB downloaded
  Running setup.py (path:/tmp/pip-dkcvbvbj-build/setup.py) egg_info for package from https://github.com/hgrecco/pyvisa-sim/zipball/master
    running egg_info
    creating pip-egg-info/PyVISA_sim.egg-info
    writing requirements to pip-egg-info/PyVISA_sim.egg-info/requires.txt
    writing pip-egg-info/PyVISA_sim.egg-info/PKG-INFO
    writing top-level names to pip-egg-info/PyVISA_sim.egg-info/top_level.txt
    writing dependency_links to pip-egg-info/PyVISA_sim.egg-info/dependency_links.txt
    writing manifest file 'pip-egg-info/PyVISA_sim.egg-info/SOURCES.txt'
    warning: manifest_maker: standard file '-c' not found
    
    error: package directory 'pyvisa-sim/testsuite' does not exist
    Complete output from command python setup.py egg_info:
    running egg_info

creating pip-egg-info/PyVISA_sim.egg-info

writing requirements to pip-egg-info/PyVISA_sim.egg-info/requires.txt

writing pip-egg-info/PyVISA_sim.egg-info/PKG-INFO

writing top-level names to pip-egg-info/PyVISA_sim.egg-info/top_level.txt

writing dependency_links to pip-egg-info/PyVISA_sim.egg-info/dependency_links.txt

writing manifest file 'pip-egg-info/PyVISA_sim.egg-info/SOURCES.txt'

warning: manifest_maker: standard file '-c' not found

error: package directory 'pyvisa-sim/testsuite' does not exist

----------------------------------------
Cleaning up...
  Removing temporary dir /tmp/pip_build_marquardt...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-dkcvbvbj-build
Exception information:
Traceback (most recent call last):
  File "/sw/python/anaconda3-2.0.1/lib/python3.4/site-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/sw/python/anaconda3-2.0.1/lib/python3.4/site-packages/pip/commands/install.py", line 278, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/sw/python/anaconda3-2.0.1/lib/python3.4/site-packages/pip/req.py", line 1229, in prepare_files
    req_to_install.run_egg_info()
  File "/sw/python/anaconda3-2.0.1/lib/python3.4/site-packages/pip/req.py", line 325, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/sw/python/anaconda3-2.0.1/lib/python3.4/site-packages/pip/util.py", line 697, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pip.exceptions.InstallationError: Command python setup.py egg_info failed with error code 1 in /tmp/pip-dkcvbvbj-build

Storing debug log for failure in /home/marquardt/.pip/pip.log

hgrecco, 2015-02-09:
Thanks Colin!. It should be fixed now. Please let me know if it works for you and close the issue if needed.

cmarqu, 2015-02-09:
Works now, thanks!

closed 2015-02-09 by cmarqu