pyvisa/pyvisa-sim #7: Parser: defining properties

hgrecco, 2015-02-09:
Syntax:
```
@property <property name>[: <specification>]
get >>> <get string>
get <<< <return format using stringparser>
set >>> <set string using PEP 3101>
[set <<< <return string when everything is ok using stringparser>]
[err <<< <return string when an error is  found using stringparser>]
```

Things between brackets are optional.
If `set <<<` or `err <<<` are not given, it is assumed that nothing is returned.
If the specification is not given, any value is possible.

For example:
```
@property frequency
get >>> FREQ?
get <<< {.3f}
set >>> FREQ {.3f}
set <<< OK
err <<< ERROR
@end 
```

#### How to write the specification

**A** using lantz-like kwargs Feat.
Range: `limits = min, max, step value`
Valid values: `values = value1, value2, value3`

**B** using a more texty way:
Range: `min < value < max`
Step: `step = step value`
Valid values: `value1 | value2 | value3`

MatthieuDartiailh, 2015-02-09:
I was wondering if it might not be possible to accept arbitrary function definition (marked with lets say '@python' to allow more exotic stuff (as it seems to me it often ends up not being that exotic). When parsing the file the def can be exec and assigned as getter or setter. We should only agree on a given signature.

hgrecco, 2015-02-09:
Sounds reasonable to me.

MatthieuDartiailh, 2015-02-09:
Great, I have some notes about that somewhere. When I find them I will post them here.

hgrecco, 2015-02-10:
I have implemented a YAML definition file to try out a few ideas. A few things are more verbose but in general is quite nice.

hgrecco, 2015-02-11:
I will open another issue for the `@python` idea

hgrecco, 2015-02-11:
Implemented using YAML,

closed 2015-02-11 by hgrecco