pyvisa/pyvisa-sim #9: Add support for instrument with channels.

MatthieuDartiailh, 2015-02-11:
I think the idea would simply be to store the parameters in different dictionaries according to the channel. Two main mechanism should supported :
- the active channel is defined through a command
- the channel is specified as part of the command (the string parser might be of help here).

hgrecco, 2015-02-11:
Agreed. Maybe the way is to add a `channels`  entry in the property dict and require that getters and setters are written with named placeholders such as `{channel:d}` and `{value:.3f}`

I will release version 0.1 later this week and start with this afterwards.

MatthieuDartiailh, 2015-02-11:
It might be better to keep track of the channels at the instrument level rather that at the property level, as often instruments have many (a dozen) property which apply to channel. Could you do some hierarchical declaration in the YAML like this:
``` yaml
device PNA:
    properties : 
         # instrument wide properties
    channel :
         properties:
              # channel specific declaration.
```
Otherwise I fine with the use of '{channel}' or '{ch_id}' (may make sense to use the same we may use in lantz, whatever it ends to be).

famish99, 2015-03-03:
I think the latter implementation is only manageable if issue #16 is resolved first, otherwise for channels that share a lot of the same properties, it could get a bit tedious to define them.

MatthieuDartiailh, 2015-05-10:
I might start working on this during this week. @hgrecco can you give any pointer where to look at first ?

famish99, 2015-05-11:
Most of the work will be in devices.py where you'll have to modify getters and setters inside of the Device class. Some work may be necessary in parser.py to accomodate the new yaml tags.

hgrecco, 2015-05-12:
Indeed most changes will be in `devices.py` @MatthieuDartiailh If you proceed to implement your hierarchy, it would be a good idea to have a way to specify the valid values fo r the channel.

MatthieuDartiailh, 2015-05-12:
It is my intention indeed. But the ideal would be to specify in the resource which channel are active. But I will let that for another PR.

hgrecco, 2015-05-20:
close as of #27

closed 2015-05-20 by hgrecco