pyvisa/pyvisa-sim #15: Allow writes with no expected response

famish99, 2015-03-02:
Most of the SCPI based equipment I use (Agilent DMM, DAQ, etc.) if the command is write-only, will not allow you to do a read afterwards. The mocks must be queried (write+read) in order to flush the newline response. There should be some way to make the writes work without having to do a read.

Current workaround is overwriting my write() function with query() for the unittests, but I'd like a way to do this more cleanly.

hgrecco, 2015-03-03:
Yes. This is an unintended behavior. Would you like to submit a patch? It could be along the following lines.

If there is not `r` field in the dialogue, store a particular sentinel object (maybe called MISSING). You will need to change at least the parser and `to_bytes`) 
Then in `Device.write`, if the matched response is this MISSING value then do not write the eom.

famish99, 2015-03-03:
I can look into it because I'm running into it quite a bit. Thanks for the pointers

hgrecco, 2015-03-06:
I think this can be closed now.

closed 2015-03-06 by hgrecco