pyvisa/pyvisa-sim #16: Allow basing a simulated device in another

hgrecco, 2015-03-03:
It is common that multiple devices will share a large part of the commands. Here, we propose a syntax to facilitate this.

We will add to the device dict a new key (`bases`) with the value specifying a list of devices used as bases. Multiple bases are allowed. Bases first in the list have precedence.

Each element of the list is dictionary specifying the device like it is done in the [resources](http://pyvisa-sim.readthedocs.org/en/latest/definitions.html#resources) section.

```
    <device name>:
        bases: 
             - device: base 1
             - device: base 2
               filename: myfile.yaml
             - device: base 3
               filename: default.yaml
               bundled: true
        <here goes the device definition>
```

#### Implementation

For each base, its dictionary is loaded and merged with the derived device.

famish99, 2015-03-30:
How well does this work? I might want to use this to take a stab at issue #9

MatthieuDartiailh, 2015-05-10:
This should be closed @hgrecco, no ?

hgrecco, 2015-05-12:
Yes. I think the implementation is good. I am still improving the parser but will not corrupt this.

closed 2015-05-12 by hgrecco