pyvisa/pyvisa-sim #32: Remove raising errors from SimVisaLibrary

hgrecco, 2015-05-19:
In order to allow an homogeneous and user customizable policy about errors and warnings the proposal is to remove raising error and logging warnings from SimVisaLibrary. Instead, these functions should return the status code (i.e a member of `constants.StatusCode` or  `constants.VI_SUCCESS_*` / `constants.VI_WARN_*` / `constants.VI_ERROR_*`).

Another layer will take care of raising the appropriate error and log warnings. An open question is how to implement this layer. I see the following options:
  1. add a decorator to the functions (maybe via a class or metaclass, to make it easier). There is a stub for such decorator [here](https://github.com/hgrecco/pyvisa/blob/master/pyvisa/errors.py#L476). 

  2. prefix all current 2 level functions with an underscore (`_`) and change VISALibraryBase to dispatch to those functions.

@famish99 @MatthieuDartiailh

hgrecco, 2015-05-19:
By the way, the same policy will be applied to pyvisa-sim and the ni backend

MatthieuDartiailh, 2015-05-19:
What bothers me is that if we do that we won't have a nice traceback to follow to determine the origin of the error. When debugging a pyvisa-sim .yaml it is invaluable ! I would be in favour of introducing a helper function in charge of adding some information to the error and then do a standard reraising.
As we only catch nicely behaved exception whose args[0] exists and is a string we should be able to do that easily. The best way would be to use raise ... from ... but this is sadly python 3 only.

hgrecco, 2015-05-19:
I think there are two types of errors. I was only referring for the ones supported by NI-VISA (like a timeout). I am fine with raising the others.

famish99, 2015-05-28:
Yeah I think there should be some work done at the parser layer to catch bugs to make debugging the yaml files easier. Also I have an intern working with me for the summer whose task is to improve our unittesting framework so I may be able to get some of these tasks completed.

On a separate note, we wish to have a framework where we can tie mocked equipment together (e.g. changing a mock power supply would show up on a mock multimeter) is this effort existing anywhere or should we create a pyvisa-system-sim side project for the task?

hgrecco, 2015-06-01:
I think that the way to facilitate debugging pyvisa-sim is failing fast, logging and decoupling the VISA status codes from the real errors.

Regarding your separate note, that is something in which we definitely interested. Before pyvisa-sim started I did something like this for Lantz (See [here](https://github.com/LabPy/lantz/tree/master/lantz/simulators)) Making something like this in a more robust and configurable way will be awesome. Open another issue to brainstorm ideas.