pyvisa/pyvisa-sim #36: Add support for query in list_resources.

MatthieuDartiailh, 2015-07-22:
Once again this is necessary for Lantz testing.

hgrecco, 2015-07-22:
Actually I have this partially implemented in pyvisa-py. My plan is to move it to `VisaLibraryBase` (in pyvisa) to allow other backend to use it or overwrite if necessary (in the same as we are doing it with `parse_resource`.  We can then continue improving the support for more complicated queries.

MatthieuDartiailh, 2015-07-22:
Good to know. I beed this to properly test the via_usb methods.

hgrecco, 2015-07-23:
Done!. There is a function in `pyvisa.rname` called `filter` that implements the filtering. It is still very basic, but will do the trick to start. We can then implement the other parts of the VISA expr language.

You need the latest develop of PyVISA for this. As soon as we test it a little bit more, I will push a new PyVISA, PyVISA-py and PyVISA-sim release.

closed 2015-07-23 by hgrecco

MatthieuDartiailh, 2015-07-23:
I should be able to add the tests to Lantz the first week of august. I will keep you posted.