pyvisa/pyvisa-sim #38: How to simulate instrument that returns data when inst.read()

DavidLutton, 2016-06-25:
I have a HP 437B that when instrument.read() returns the measured power value that can be interpreted as a float.

I read the [pyvisa-sim definitions](https://pyvisa-sim.readthedocs.io/en/latest/definitions.html#definitions) guide and did not find where I could add a property that returns on a query-less .read()

My sim [HP-437b.yaml](https://gist.github.com/DavidLutton/ff6d5455f8e98d92ac12f002d69fbc7a) so far

[437B-read.py](https://gist.github.com/DavidLutton/3f5a415a120a81cdc37d89737f357bbc) to demonstrate how it currently works with the expected behaviour commented out on the last line

Can we add this behaviour when there is no query?
getter:
  q: None

MatthieuDartiailh, 2016-06-25:
Just to be sure : at any moment if you read the instrument buffer using instrument.read() you get the measured power, without sending any message first ? 

So if you do : 
````python
instrument.write('*IDN?')
instrument.read() # you get the idn
instrument.read() # you get the measured power
instrument.read() # you get the power again 
```
you get the results suggested by the comments ?

DavidLutton, 2016-06-25:
Yes typically, other than times where the write has not been processed by the instrument before the first read is occurring.  
```
instrument.write('*IDN?')
instrument.read() # you get the measured power
instrument.read() # you get the idn
instrument.read() # you get the power again 
instrument.read() # you get the power again 
instrument.read() # you get the power again 
instrument.read() # you get the power again 
```
2591 * True copied data back out of spreadsheet and code at bottom of [Gist](https://gist.github.com/DavidLutton/43e1b787ccb961c63d601afb630a3787) 
# Delta between runs; yes I was saving the file inside the loop each time

MatthieuDartiailh, 2016-06-27:
This a pretty unusual behavior and I do not think that pyvisa-sim currently supports it. (yet)

DavidLutton, 2016-06-27:
Good news, tried again with the hardware
Don't seem to get the order of response glitch, might of been in a menu when I encountered that

They do work the way I said but also they respond to instrument.query('?') with the measured power
I read the 'HB-IP codes summery' list and '?' is not listed

Thanks for your help

closed 2016-06-27 by DavidLutton