pyvisa/pyvisa-sim #49: Random values as response

lucabecchetti, 2020-02-05:
Is is possibile do define a query that return random values, for example random string or radom number?

for example:

- q: "TRIGGER:A:MODE?"
   r: RANDOM("AUTO", "NORMAL", "ETC")

Thanks!

lucabecchetti, 2020-08-06:
up

murvar, 2022-11-16:
I'm new to the world of open source software but I'll give this a go! 

I've tried to implement this functionality by checking if a setter includes the substring RANDOM. It works on all three types of properties - strings, floats and integers. 
Randomizing a string property sets a value from the valid specs list.
Randomizing a float/integer sets a value between the min and max specs. 

Would this solve your issue?

You can try it out by swapping the component.py file for the one I've included here.

[RandomComponent.zip](https://github.com/pyvisa/pyvisa-sim/files/10020720/RandomComponent.zip)