pyvisa/pyvisa-sim #52: pyvisa doesn't found pyvisa-sim with pyvisa 1.11.0

joseph2rs, 2020-09-21:
With the new PyVisa 1.11.0, pyvisa is not able to load pyvisa-sim backend.

```python
import visa

rm = visa.ResourceManager('@sim')
inst = rm.open_resource('ASRL1::INSTR', read_termination='\n')
```

Stacktrace :

```
>       rm = visa.ResourceManager('@sim')

visa_wrapper.py:21: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
venv/lib/python3.6/site-packages/pyvisa/highlevel.py:3005: in __new__
    visa_library = open_visa_library(visa_library)
venv/lib/python3.6/site-packages/pyvisa/highlevel.py:2919: in open_visa_library
    return cls(argument)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

cls = <class 'pyvisa-sim.highlevel.SimVisaLibrary'>, library_path = ''

    def __new__(
        cls: Type[T], library_path: Union[str, LibraryPath] = ""
    ) -> "VisaLibraryBase":
        """Create a new VISA library from the specified path.
    
        If a library was already created using the same path and class this
        library object is returned instead.
    
        Parameters
        ----------
        library_path : str | LibraryPath
            Path to the VISA library to use in the backend.
    
        Raises
        ------
        OSError
            Raised if the VISA library object could not be created.
    
        """
        if library_path == "":
            errs = []
            for path in cls.get_library_paths():
                try:
                    return cls(path)
                except OSError as e:
                    logger.debug("Could not open VISA library %s: %s", path, str(e))
                    errs.append(str(e))
                except Exception as e:
                    errs.append(str(e))
            else:
>               raise OSError("Could not open VISA library:\n" + "\n".join(errs))
E               OSError: Could not open VISA library:

venv/lib/python3.6/site-packages/pyvisa/highlevel.py:173: OSError
```

It works as expected with 1.10.1

Installed versions :

pyvisa==1.11.0
pyvisa-sim==0.3

Ubuntu 18.04 (x64)
Python 3.6 (x64)

No other VISA backends installed

MatthieuDartiailh, 2020-09-21:
Sorry about that. I do not actively maintain this package since I was unconvinced from the start by some design choices, but I was not planning to break it. The issue seems to be related to imports from `pyvisa.compat` if you track them down, everything should work again. If you can submit a PR it would be great.

jscoughlin, 2020-10-06:
I haven't dug into this too much, so sorry if I'm just repeating the same issue, but I notice that with pyvisa 1.11.0 (and 1.11.1) I'm unable to use pyvisa-sim even with my custom yaml. As with @joseph2rs, I don't have any problems with 1.10.0.

The issue seems related to the "write_raw" method in pyvisa/messagebased.py.

`self = <'SerialInstrument'('ASRL13::INSTR')>, message = b'SYSTEM:REMOTE\r\n'

    def write_raw(self, message: bytes) -> int:
        """Write a byte message to the device.
    
        Parameters
        ----------
        message : bytes
            The message to be sent.
    
        Returns
        -------
        int
            Number of bytes written
    
        """
>       return self.visalib.write(self.session, message)[0]
E       TypeError: 'NoneType' object is not subscriptable`

MatthieuDartiailh, 2020-10-06:
Pyvisa-sim relies on the presence of pyvisa.compat which was removed with the end of support for Python 2.  I am kind of surprised the system does not crash earlier... I will do my best to put a PR tomorrow night to address this issue.

closed 2020-10-15 by MatthieuDartiailh

MatthieuDartiailh, 2020-10-15:
0.4.0 is released and addresses this issue.

jscoughlin, 2020-10-16:
Thanks @MatthieuDartiailh. 0.4.0 fixed my problem!

PR #53 (2020-10-15): Drop python 2 support and support Pyvisa >= 1.11
Should close #52 

Please test this pass the tests but I do not have anything more involved to test it on.