pyvisa/pyvisa-sim #62: Conda installs version 0.3 -- Get no package pyvisa_sim found.

nate-h, 2022-10-29:
Hi, would love to use this package.
I'm using python version 3.9 w/ conda.
My env list shows version 0.3 is installed.
When I run:

```py
>>> import pyvisa
>>> pyvisa.ResourceManager("@sim")
```

I get `ValueError: Wrapper not found: No package named pyvisa_sim`

If I uninstall then use pip to install, it downloads version 0.5.1 and that code works.

Any chance this can be fixed? Would love to help if I can.

MatthieuDartiailh, 2022-10-29:
There is currently no feedstock for pyvisa-sim. You can make a PR to add one and feel free to add me as maintainer.

MatthieuDartiailh, 2022-11-02:
The issue you opened is in pyvisa-py feedstock we will need a different feedstock for pyvisa-sim.

mtsolmn, 2022-11-27:
@MatthieuDartiailh this issue can be closed since we've added `pyvisa-sim` to `conda-forge`: https://github.com/conda-forge/pyvisa-sim-feedstock

thanks

MatthieuDartiailh, 2022-11-27:
Thanks @mtsolmn

closed 2022-11-27 by MatthieuDartiailh