pyvisa/pyvisa-sim #68: multiple instances of ResourceMaganer

yn4k4nishi, 2022-12-15:
when I create multiple instance of ResourceManager, dialogue response is wrong.
Is it not recommended?

my environment:
- python 3.8.10
- PyVISA                        1.11.3   
- PyVISA-py                     0.5.2    
- PyVISA-sim                    0.4.0

MatthieuDartiailh, 2022-12-15:
Could you provide a sample script illustrating your issue ?

bilderbuchi, 2023-01-25:
Not sure what @yn4k4nishi means, but I also observe weird problems running pytest tests with pyvisa-sim backed fixture objects, in that replies remain in a read buffer despite the objects + resource managers being destroyed:
```python
>>> import pyvisa
>>> rm = pyvisa.ResourceManager(visa_library='@sim')
>>> instr = rm.open_resource('ASRL2::INSTR')
>>> instr.write('*IDN?')
7
>>> instr.read()
'SCPI,MOCK,VERSION_1.0\n'  # OK, that's what I expect
>>> instr.write('*IDN?')  # we don't fetch that response from the instrument
7
>>> id(instr)
140011221241280
>>> del instr  # Kill the instrument
>>> del rm
>>> rm = pyvisa.ResourceManager(visa_library='@sim')
>>> instr = rm.open_resource('ASRL2::INSTR')
>>> instr.read()  # Why is this response in this different instrument object?
'SCPI,MOCK,VERSION_1.0\n'
>>> id(instr)
140011221241472  # the IDs of the instr objects are different, so it's not somehow the same
``` 
Is that expected?

MatthieuDartiailh, 2023-01-25:
Here you are assuming that when you delete the instr and the resource manager they instantaneously go away. This may not be true because the rm and the library object are in a reference cycle and the gc may dispose of them only later on. Could you run the same code but explicitly closing the instr and rm rather than relying on `__del__` being invoked.

bilderbuchi, 2023-01-25:
Ah, you are right. `instr.close()` is neither sufficient nor necessary, `rm.close()` does the trick. Thanks!

MatthieuDartiailh, 2023-01-25:
The fact that closing the instrument does not solve the issue still says that something is off. Instruments do not share outgoing buffers between connections and we should mimic that.

closed 2023-04-13 by yn4k4nishi

bilderbuchi, 2023-04-28:
> The fact that closing the instrument does not solve the issue still says that something is off. Instruments do not share outgoing buffers between connections and we should mimic that.

@MatthieuDartiailh is that something that still should be followed up separately? (to avoid this disappearing in a closed issue comment)

MatthieuDartiailh, 2023-04-28:
It would make sense yes