pyvisa/pyvisa-sim #93: bytes_in_buffers does not work in simulation

LongnoseRob, 2023-12-26:
I am woking on a testbench with pyvisa-sim, and I found with the sim backend, `instr.bytes_in_buffer` for a simulate serial conenction does not work:

```
$python builds/locl/sim/sim_test.py 
LSG Serial #1234
Traceback (most recent call last):
  File "/home/robby/builds/locl/sim/sim_test.py", line 17, in <module>
    print(instr.bytes_in_buffer)
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/robby/builds/pyvisa/pyvisa/attributes.py", line 175, in __get__
    return self.post_get(instance.get_visa_attribute(self.attribute_id))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/robby/builds/pyvisa/pyvisa/attributes.py", line 286, in post_get
    return int(value)
           ^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'object'
```
minimalist setup to reproduce:
[sim_test.py.txt](https://github.com/pyvisa/pyvisa-sim/files/13768053/sim_test.py.txt)

Is there an easy way to have this functional in pyvisa-sim?

MatthieuDartiailh, 2023-12-28:
@LongnoseRob I looked into the issue and it appears that pyvisa-sim handling of attributes is quite limited ATM. So the first things to do will be to expand it to support more complex cases (custom getters and setters). 

I will add it to my todo list but feel free to beat me to it if you have the bandwidth.

LongnoseRob, 2024-01-03:
@MatthieuDartiailh I was thinking about "inregtrating" the serial resource from pyvisa-py (which supports pyserials loop://) into the serial  resource part of -sim, but have not yet had a deep look into where and how to do it.
A mixin comes to mind, but I am not sure if this would be the correct way.

HNY BTW