pyvisa/pyvisa-sim #167: Regression: can no longer set a query-without-response in the `dialogues` section

dougthor42, 2025-09-02:
ef0c60c76458697368ae8a6a3cfd4bdb1b25478a caused dialog query-without-response to no longer work:

```yaml
dialogues:
  - q: "foobar"
```

Thus far I've only seen the issue with the "dialogues" section; I haven't tested other sections.

### To test:

1.  `cp ./pyvisa_sim/testsuite/fixtures/channels.yaml ./my_channels.yaml`
2.  Edit `my_channels.yaml`:
    ```diff
    --- a/pyvisa_sim/testsuite/fixtures/channels.yaml
    +++ b/my_channels.yaml
    @@ -18,6 +18,7 @@ devices:
         dialogues:
           - q: "?IDN"
             r: "LSG Serial #1234"
    +      - q: "foobar"
         properties:
           selected_channel:
             default: 1
    ```
3.  Make a script to run during the `git bisect`:
    ```python
    #!/usr/bin/python3
    import pathlib
    
    import pyvisa
    
    path = pathlib.Path(__file__).parent / "my_channels.yaml"
    value = "foobar"
    try:
        rm = pyvisa.ResourceManager(f"{path}@sim")
    
        instr = rm.open_resource(
            "USB::0x1111::0x2222::0x1234::INSTR",
            read_termination="\n",
            write_termination="\n",
        )
        got = instr.write(value)
    finally:
        rm.close()
    assert got == len(value) + 1  # value + the write_termination
    ```
4.  Run `git bisect`, checking that script on each commit.
    +   Result: ef0c60c76458697368ae8a6a3cfd4bdb1b25478a
    +   N.B.: I didn't describe other minor requirements such as installing pyvisa_sim as an editable pip package or setting up the script in such a way that it can be run by `git bisect run`.

dougthor42, 2025-09-02:
Oh, I forgot to mention in the OP that the test case that found this in our code, when we upgraded from pyvisa_sim 0.5.1 to 0.7.0 was this:

```
# sim file:
dialogues:
  - q: "*RST"

# python test
assert instr.write("*RST") == 6  # 4 chars + 2 write_termination
```

dougthor42, 2025-09-02:
I've confirmed that removing this line:

```diff
--- a/pyvisa_sim/component.py
+++ b/pyvisa_sim/component.py
@@ -311,9 +311,6 @@ class Component:
             response = dialogues[query]
             logger.debug("Found response in queries: %s" % repr(response))
 
-            if "RANDOM" in response.decode("utf-8"):
-                response = random_response(response.decode("utf-8")).encode("utf-8")
-
             return response
 
         return None
```

resolves things. Now to figure out why...

MatthieuDartiailh, 2025-09-02:
Thanks for the report. I will have a look.

dougthor42, 2025-09-02:
Oh it looks like you've already applied the same fix I was about to suggest, thanks!

PR #168 (2025-09-02): Fix support for response less dialog that was broken by support for RANDOM
Closes #167